witnessmenow/Universal-Arduino-Telegram-Bot #319: multiple bot tokens? (one for status, one for alerts)

quilmore, 2023-02-20:
I've been succesfull on doing all I needed but I'd like to have a bot for status that I can mute in telegram and check every now and then (it sends notifications of events as they happen) and another bot for alarms that could notify me on the phone when something needs attention.
How can that be done?

As far as I could test, this line can only happen once and not be changed in loop:
UniversalTelegramBot bot(BOT_TOKEN, secured_client);

I wouldn't mind changing the destination for an alarm and setting it back to the normal bot for everything else
or can I include the bot I need to send to in the bot.sendMessage?

Hegy, 2023-02-20:
Hello! May I give you a piece of advice? You can youse one bot to write to the different groups. Create a group or channel for the different types of notification. Add your bot to this group and tell him to write there. That will give you the opportunity to mute that group.

quilmore, 2023-02-20:
> Hello! May I give you a piece of advice? You can use one bot to write to the different groups. Create a group or channel for the different types of notification. Add your bot to this group and tell him to write there. That will give you the opportunity to mute that group.

Thanks Hegy,
Let me see if I understand you correctly, add the bot to 2 groups lets say, STATUS_GROUP and ALERT_GROUP, then use the CHAT_ID of each group to decide what to send and have the STATUS_GROUP muted and the ALERT_GROUP not muted so I get the alarms? 
Actually by typing the question I think that it will work, thanks, will try. 
But if multiple bots could be used I'll be interested to know and willing to test any development.

quilmore, 2023-02-20:
> I think that it will work

It does work! tricky to find the CHAT_ID of a group and threw me out that it was a negative number but I've got it to work, thanks
I'll leave the issue open to see if there is a possibility to manage more than 1 bot at the time

witnessmenow, 2023-04-01:
You can swap out bot tokens if you want to run multiple bots at once I guess!

```
String bot1 = "abc1234";
String bot2 = "xyz9876";

bot.updateToken(bot1);

bot.sendMessage(CHAT_ID, "Will be sent as bot 1", "");

bot.updateToken(bot2);

bot.sendMessage(CHAT_ID, "Will be sent as bot 2", "");

```

closed 2023-04-01 by witnessmenow