witnessmenow/Universal-Arduino-Telegram-Bot #333: How to not receive the same message again after restart?

dmikushin, 2023-07-11:
I'm doing `ESP.restart()` in response to a restart message sent by Telegram client. However, after device reboots, it get the same restart message, and eventually restarts forever. How do I avoid executing the same message again after restart?

Dilbao, 2023-07-26:
If you restart from within the function that processes messages, the processing of the current message will be incomplete. Instead do something like this:
- Make a global variable: `bool RestartTriggered = false;`
- Change the variable instead at where the `ESP.restart()` is currently: `RestartTriggered = true;`
- Add a check in your `loop()` for restart: `if (RestartTriggered == true) {ESP.restart();}`

robedua, 2024-04-27:
Boa tarde isso não funciona !!!!!

ldijkman, 2024-05-11:
Hmmm
think i doing it as describe
but i get a boot loop

https://t.me/Luberth_Dijkman
https://github.com/ldijkman/async-esp-fs-webserver/blob/master/docs/ino/thermostat_web_flash/chatgpt_thermostat.ino

[image]