witnessmenow/Universal-Arduino-Telegram-Bot #336: write / respond to a group

goguelnikov, 2023-09-01:
Hello,
Amazing work Brian, thank you so much! The library is really nice and easy to use!
I have a question:
I can have the bot to get the message / reply when I chat with it directly one to one.
I'd like my bot to be part of a Telegram group and be able to get the message from this group and respond to this group.
The ESP32 only get triggered when I put "/" at the beginning of the message in a group... and in this case, it does not reply to the group…
Do you have any idea?
Best

Hegy, 2023-09-01:
Hello!

So if you put a code like this you won't get any answer like I do?
`    if (text == "test message") {
        bot.sendMessage(chat_id, "Test message recieved", "");
    }`

Please consider to share your code. At least a function which handleNewMessages

[image]

goguelnikov, 2023-09-01:
here is my code 

void handleNewMessages(int numNewMessages) {
  Serial.println("handleNewMessages");
  Serial.println(String(numNewMessages));

  for (int i=0; i<numNewMessages; i++) {
    // Chat id of the requester
    String chat_id = String(bot.messages[i].chat_id);
    if (chat_id != CHAT_ID){
      bot.sendMessage(chat_id, "Unauthorized user", "");
      continue;
    }
    
    // Print the received message
    String text = bot.messages[i].text;
    Serial.println(text);
    String from_name = bot.messages[i].from_name;

  }
}

goguelnikov, 2023-09-07:
No idea?
Can any body help?

Hegy, 2023-09-15:
> No idea? Can anybody help?

Just o double check: Do you have a call of handleNewMessages() from void loop() ?

And again! Add this code in your  **handleNewMessages** function

`    if (text == "test message") {
        bot.sendMessage(chat_id, "Test message recieved", "");
    }`

goguelnikov, 2023-09-15:
I used the standard handleNewMessages
you can check the code here : https://github.com/goguelnikov/TelegramMinitelChat/blob/main/src/main.cpp
the part : 
    if (chat_id == CHAT_ID){
      bot.sendMessage(chat_id, text, "");
    } else {
      bot.sendMessage(chat_id, "Unauthorized", "");
    }
is commented
the handleNewMessages is indeed inside void loop()
I'll try you're coode (but I think I did already)

goguelnikov, 2023-09-18:
hello
here is my code: 
void handleNewMessages(int numNewMessages) {
  //Serial.println(F("handleNewMessages"));
  Serial.print(F("NB message: "));
  Serial.println(String(numNewMessages));

  for (int i=0; i<numNewMessages; i++) {
    // Chat id of the requester
    Serial.print(F("From Chat_ID: "));
    String chat_id = String(bot.messages[i].chat_id);
    Serial.println(chat_id);
    Serial.print(F("Message: "));
    String text = bot.messages[i].text;
    Serial.println(text);
    Serial.print(F("From_name: "));
    String from_name = bot.messages[i].from_name;

    if (chat_id == CHAT_ID){
       bot.sendMessage(chat_id, text, "");
    } else {
       bot.sendMessage(chat_id, "Unauthorized", "");
    }
  }
}

here is my complete analysis :
If I put the bot in a group with group ID XYZ (Chat_ID)
- if I send a message to from this group : no reply (no log on the serial monitor)
- if I send a message to from another group : no reply (no log on the serial monitor)
- if I send a message directly to the bot (one to one), here I get a reaction
      From Chat_ID: XXXXXXX
      Message: hello
      From_name:
and a message "Unauthorized"

If I chat directly withe the bot (Chat_ID = bot_id)
- the bot reacts to message and can reply and send message to me...

Hegy, 2023-09-19:
Have you disable Group Privacy in your bot settings?

go to [@BotFather](https://t.me/BotFather)
send /mybots
select your bot
click Bot Settings
click Group Privacy (it should go like "Privacy mode is **disabled** for your_bot_name_bot")

**REMOVE** your bot from the group
Add your bot to the group

goguelnikov, 2023-09-19:
yeah !!! it seems to be working ! nice !
this was the problem
thank you !

closed 2023-09-24 by goguelnikov