witnessmenow/Universal-Arduino-Telegram-Bot #337: Retrieving time

ehraz786, 2023-09-24:
is retrieving time from ntp necessary?
without "time_t now" the bot doesn't receive any message.

```
Serial.print("Retrieving time: ");
  configTime(0, 0, "pool.ntp.org");
  time_t now = time(nullptr);
  while (now < 24 * 3600)
  {
    Serial.print(".");
    delay(100);
    now = time(nullptr);
  }
  Serial.println(now);
```

Hegy, 2023-09-29:
Hello!
Please try to connect to another wifi device (router or hot-spot on your mobile).

ehraz786, 2023-09-29:
Tried it.
Without retrieving the time, the bot fails to receive any message.

Hegy, 2023-09-29:
Is your bot able to send a message without retrieving the time?

ehraz786, 2023-09-30:
I am using the example sketch SendMessageFromEvent.ino from examples/ESP8266/SendMessageFromEvent
```
bot.sendMessage(CHAT_ID, "Bot started up", "");
```
This line works well if this 
```
Serial.print("Retrieving time: ");
  configTime(0, 0, "pool.ntp.org"); // get UTC time via NTP
  time_t now = time(nullptr);
  while (now < 24 * 3600)
  {
    Serial.print(".");
    delay(100);
    now = time(nullptr);
  }
  Serial.println(now);
```
but fails to do anything upon removing it

dannyz2, 2025-04-01:
I realize this is a very old thread, but, someone may read it & still need the answer. The "client.setCACert(TELEGRAM_CERTIFICATE_ROOT);" requires the time to be able to authenticate.