witnessmenow/Universal-Arduino-Telegram-Bot #341: datatype difference : last_message_received (long) and update_id (int)

ceewanna, 2024-01-04:
Under "processResult" last_message_received takes value from update_id. 

Is there any rationale behind having different datatype for these 2 variables?

witnessmenow, 2024-01-04:
I guess it's a bug, not sure which one it should be to be honest! probably long?

witnessmenow, 2024-01-04:
changed in #330 actually

ceewanna, 2024-01-05:
Thanks. 

By the way I am exploring ways to get multiple devices taking commands from one chat bot. Every device has its own id number and the command is attached with the id. Every device reading from the chat bot and only act on the command having id matched.

I am not so sure if by calling getUpdates the last_message_received will be moving on and some of the devices will miss its command. So I have each device maintaining its own tracking of last_message_received (through update_id). At the start each device will call getUpdates(bot.last_message_received+1) to get the proper bot.last_message_received value and make use of its as a starting tracking of message. Every bot.messages[i] each device processes it will update the tracking from update_id. Subsequently the getUpdaes will be callled based on each tracking+1.

Please correct me if I am wrong about the use of last_message_received. Do you have any guidance on how to handle the multiple devices in my case?

witnessmenow, 2024-01-05:
This is a path filled with pitfalls :) technically possible, but messy. If one of your devices go down and their is a message waiting for it, the messages will stop coming.

I would use it as a last resort. Multiple bots in a group might be a better option for example.

I think from reading your post that your idea of how the messageId thing works is wrong, 

Say your last actually messaged ID was 100

When you send a `getUpdates(100+1)` from any device, Telegram will delete any message it finds with a lower ID than 101 from its server as it considers them handled. 

That means no device will see a message with a lower ID than 101 from this point on, even if they send `getUpdates(80)`, it will not return any message with a lower ID than 101

How to implement: So in your scenario, you will have a few different boards, lets say they are labeled A, B and C

You send a message intended for C, so maybe your include the label in the message "/on C" or something.

you would need to extract the label and check them against your board.

Psuedo code:

Make messageIdToSend a global variable, you can default it to 0
```
long messageIdToSend = 0; //or int if it should be int
```

in your method for processing the message
```
String label = extractLabelFromMessage(message.text);
if(label == this_device_label){
     //Act on message (turn on your light or whatever)
     messageIdToSend = message.id +1
} else {
    //Don't do anything, messageIdToSend doesn't need to be updated, the device the message it is for should delete it on Telegrams server when it gets it

//Maybe you could keep a counter here that if your receive the same message.id several times in a row it means the board the message is for is probably not working, and you could unblock the queue
}

```

and then send messageIdToSend in your loop when calling getUpdates
```
getUpdates(messageIdToSend )
```