witnessmenow/Universal-Arduino-Telegram-Bot #345: arduino uno r4 wifi weirdly stucks after a couple of hours

berthelotito, 2024-02-02:
Hi to everyone and thanks for all your hard work to do this library (mainly witnessmenow).

I've tried to modify the code for using it on my arduino uno rev4 Wifi.

I'm able to use it during a couple of hours before the code stucks.

I've been checking all the issues and libraries, but I've no success and I'm desolate :(

Regarding my research of my issue, I could check that after a couple of hours, the getUpdates() (within UniversalTelegramBot.cpp) doesn't run clientClose(); as do all the time before. This is very weird and I don't know why, because it did during more than one hour (I've tried several time and same results).

Some of the log:
when it works:
11:11:54.353 -> GET Update Messages
11:11:54.353 -> [BOT]Connecting to server
11:11:55.094 -> sending: bot.../getUpdates?offset=70732600&limit=1
11:11:56.696 -> Received empty string in response!
11:11:56.742 -> Closing client
11:11:56.789 -> GET Update Messages
11:11:56.789 -> [BOT]Connecting to server
11:11:57.483 -> sending: bot.../getUpdates?offset=70732600&limit=1
11:11:59.113 -> Received empty string in response!
11:11:59.113 -> Closing client

when it doesn't work:
11:14:12.401 -> GET Update Messages
11:14:12.401 -> [BOT]Connecting to server
11:14:13.243 -> sending: bot.../getUpdates?offset=70732600&limit=1
11:14:30.177 -> Received empty string in response!
11:14:30.232 -> GET Update Messages
11:14:30.247 -> [BOT]Connecting to server
11:14:40.287 -> [BOT]Conection error
11:14:40.337 -> Received empty string in response!
11:14:40.359 -> GET Update Messages
11:14:40.397 -> [BOT]Connecting to server
11:14:41.147 -> sending: bot.../getUpdates?offset=70732600&limit=1
11:14:57.175 -> Received empty string in response!
11:14:57.223 -> GET Update Messages
11:14:57.254 -> [BOT]Connecting to server
11:15:07.286 -> [BOT]Conection error
11:15:07.342 -> Received empty string in response!

Please any help???

Thanks,

witnessmenow, 2024-02-02:
While this library is intended for use with any board architecture, it's memory usage is geared towards what's available for the esp devices. It's a guess but I presume your running out of heap memory. You should be able to print to the serial your heap memory (not sure how on r4 wifi) and see does it trend downwards

berthelotito, 2024-02-03:
@witnessmenow thanks for your response and time helping me.

How does the esp manage the memory? Or where I could check it on the code?

I tried to print the free memory (freeMemory()) on the serial and the value was the same all the time... Anyway, I will continue doing more test.

berthelotito, 2024-02-03:
> While this library is intended for use with any board architecture, it's memory usage is geared towards what's available for the esp devices. It's a guess but I presume your running out of heap [...]

@witnessmenow thanks for your response and time helping me.

How does the esp manage the memory? Or where I could check it on the code?

I tried to print the free memory (freeMemory()) on the serial and the value was the same all the time... Anyway, I will continue doing more test.