witnessmenow/Universal-Arduino-Telegram-Bot #346: sendMultipartFormData not sending

arthur1894, 2024-02-11:
Dear friends,
For my project, I am recording audio on esp32, cutting it in smaller pieces and sending it to my computer using Universal-Arduino-Telegram-Bot. I have a problem when sending sendMultipartFormDataToTelegram in ESP32.
I have correctly set up my Bot and I am able to send messages and images, but when I send a file, nothing gets sent. The purpose is to actually send txt files, and I added the part with the cat.jpg for debugging. I am using the version of WifiClientSecure of from the esp32-1.0.4 as reccommended [here](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/128). Am i missing something here? Please recommend further steps of action to resolve this problem. Much appreciated!
Arthur

Here is a boiled down snippet of my code:

```
#include "FS.h"
#include "SD.h"
#include <WiFi.h>
#include <wificlient/WiFiClientSecure.h>    //I had to use the version from the esp32-1.0.4 package on GitHub. See https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/128
#include <telegram/UniversalTelegramBot.h>
#include <telegram/ArduinoJson.h>
#include <SPI.h>

#define WIFI_SSID "CENSORED"
#define WIFI_PASSWORD "CENSORED"
// Telegram BOT Token (Get from Botfather)
#define BOT_TOKEN "CENSORED"

#define CHAT_ID "CENSORED"

WiFiClientSecure secured_client;

UniversalTelegramBot bot(BOT_TOKEN, secured_client);

bool isMoreDataAvailable();
byte getNextByte();
bool isMoreDataAvailable()
{return sendFile.available();}
byte getNextByte()
{return sendFile.read();}
bool isMoreDataAvailable2();
byte getNextByte2();
bool isMoreDataAvailable2()
{return myFile.available();}
byte getNextByte2()
{return myFile.read();}

void send() {

String path = "/audio.txt";
File sendFile = SD.open(path, FILE_READ);
          if(!sendFile) {
            Serial.println("Failed to open sendfile for sending");  
          }
          else{
          Serial.println("sendfile present");
          }

Serial.println("**Sending audiofile.txt**");
bool sent = 
bot.sendMultipartFormDataToTelegram("sendDocument","document","audio.txt","",CHAT_ID,sendFile.size(),isMoreDataAvailable,getNextByte,nullptr,nullptr);

String file_name = "/cat.jpg";
myFile = SD.open(file_name);

Serial.println("**Sending cat.jpg**");
bool sent2 = bot.sendPhotoByBinary(CHAT_ID, "image/jpeg", myFile.size(), isMoreDataAvailable2, getNextByte2, nullptr, nullptr);

...
}
```
Here is my Serial output:

```
**Sending audiofile.txt**
[BOT Client]Connecting to server
Content-Length: 203567
Start request: --------------------------b8f610217e83e29b
content-disposition: form-data; name="chat_id"

6966033373
--------------------------b8f610217e83e29b
content-disposition: form-data; name="document"; filename="audio.txt"
Content-Type:

Sending photo by binary
End request:
--------------------------b8f610217e83e29b--
Closing client

**Sending cat.jpg**
sendPhotoByBinary: SEND Photo
[BOT Client]Connecting to server
Content-Length: 39355
Start request: --------------------------b8f610217e83e29b
content-disposition: form-data; name="chat_id"

6966033373
--------------------------b8f610217e83e29b
content-disposition: form-data; name="photo"; filename="img.jpg"
Content-Type: image/jpeg

Sending photo by binary
Sending binary photo full buffer
Sending binary photo full buffer
Sending binary photo full buffer
Sending binary photo full buffer
Sending binary photo full buffer
Sending binary photo full buffer
Sending binary photo full buffer
Sending binary photo full buffer
Sending binary photo remaining buffer
End request:
--------------------------b8f610217e83e29b--

{"ok":true,"result":{CENSORED}]}}

Closing client
{"ok":true,"result":{CENSORED}]}}

```

arthur1894, 2024-02-11:
Hi! I have completely rewritten the code and for some magical reason, it now works. Unfortunately, i cannot say where the problem lies. Thank you!

vavokado, 2024-07-13:
> Hi! I have completely rewritten the code and for some magical reason, it now works. Unfortunately, i cannot say where the problem lies. Thank you!

please share with me your codes please. my project is same like you.. please it is so important.. thank you in advance

vavokado, 2024-07-13:
> Hi! I have completely rewritten the code and for some magical reason, it now works. Unfortunately, i cannot say where the problem lies. Thank you!

vavokado@hotmail.com