witnessmenow/Universal-Arduino-Telegram-Bot #347: ArduinoJson 7 - DynamicJsonDocument is deprecated

francwalter, 2024-03-08:
With the actual ArduinoJson 7.0.3 I get, when compiling, the warning:

`warning: 'DynamicJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]`

when I downgrade to ArduinoJson 6.21.5 the warning is gone. So the use of DynamicJsonDocument should be replaced by JsonDocument.
At the moment I downgraded my ArduinoJson to 6.21.5 but I guess it is not the best solution, I prefer to keep my libraries up to date.

jameszah, 2024-04-16:
You just have to take out the size of the structure - simple but lots of typing for the library master.

https://arduinojson.org/v7/revisions/

[image]

woodlist, 2024-04-24:
In fact, newer ESP32 core and newer ArduinoJSON causes too many errors: Some of them are listed below:
`c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:40:3: note:   template argument deduction/substitution failed:
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp: In substitution of 'template<class T> ArduinoJson::V704PB2::JsonDocument::JsonDocument(const T&, ArduinoJson::V704PB2::Allocator*, typename ArduinoJson::V704PB2::detail::enable_if<(((((ArduinoJson::V704PB2::detail::is_same<T, ArduinoJson::V704PB2::JsonVariant>::value || ArduinoJson::V704PB2::detail::is_same<T, ArduinoJson::V704PB2::JsonVariantConst>::value) || ArduinoJson::V704PB2::detail::is_same<T, ArduinoJson::V704PB2::JsonArray>::value) || ArduinoJson::V704PB2::detail::is_same<T, ArduinoJson::V704PB2::JsonArrayConst>::value) || ArduinoJson::V704PB2::detail::is_same<T, ArduinoJson::V704PB2::JsonObject>::value) || ArduinoJson::V704PB2::detail::is_same<T, ArduinoJson::V704PB2::JsonObjectConst>::value)>::type*) [with T = int]':
c:\Users\woodl\Documents\Arduino\libraries\UniversalTelegramBot\src\UniversalTelegramBot.cpp:802:40:   required from here
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:40:3: error: no type named 'type' in 'struct ArduinoJson::V704PB2::detail::enable_if<false, void>'
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:3: note: candidate: 'ArduinoJson::V704PB2::JsonDocument::JsonDocument(ArduinoJson::V704PB2::JsonDocument&&)'
   33 |   JsonDocument(JsonDocument&& src)
      |   ^~~~~~~~~~~~
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:31: note:   no known conversion for argument 1 from 'int' to 'ArduinoJson::V704PB2::JsonDocument&&'
   33 |   JsonDocument(JsonDocument&& src)
      |                ~~~~~~~~~~~~~~~^~~
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:28:3: note: candidate: 'ArduinoJson::V704PB2::JsonDocument::JsonDocument(const ArduinoJson::V704PB2::JsonDocument&)'
   28 |   JsonDocument(const JsonDocument& src) : JsonDocument(src.allocator()) {
      |   ^~~~~~~~~~~~
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:28:36: note:   no known conversion for argument 1 from 'int' to 'const ArduinoJson::V704PB2::JsonDocument&'
   28 |   JsonDocument(const JsonDocument& src) : JsonDocument(src.allocator()) {
      |                ~~~~~~~~~~~~~~~~~~~~^~~
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:24:12: note: candidate: 'ArduinoJson::V704PB2::JsonDocument::JsonDocument(ArduinoJson::V704PB2::Allocator*)' (near match)
   24 |   explicit JsonDocument(Allocator* alloc = detail::DefaultAllocator::instance())
      |            ^~~~~~~~~~~~
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:24:12: note:   conversion of argument 1 would be ill-formed:
c:\Users\woodl\Documents\Arduino\libraries\UniversalTelegramBot\src\UniversalTelegramBot.cpp:802:24: error: invalid conversion from 'int' to 'ArduinoJson::V704PB2::Allocator*' [-fpermissive]
  802 |   JsonDocument payload(maxMessageLength);
      |                        ^~~~~~~~~~~~~~~~
      |                        |
      |                        int`