witnessmenow/Universal-Arduino-Telegram-Bot #366: how to ignore previous messages when ESP32/8266 boots

mjtrini, 2024-12-30:
Hello,

Thank you for your fantastic library! Is there a way to ignore/discard messages sent to the bot before the ESP boots? For example if it was not powered and a message is sent to the bot, can the message be discarded when the ESP connects to the telegram network?

This is important to prevent a string of responses when the ESP boots. I just need the ESP to respond to messages only when it has connected.

Much thanks in advance.
Michael.

witnessmenow, 2024-12-31:
I think it should be possible. What I would do is:
- On start-up do a getUpdates request as standard, this will update the
value for bot.last_message_recieved:
bot.getUpdates(bot.last_message_received + 1);
- if you get a message, repeat the getUpdates but this time pass in a
really high number: bot.getUpdates(bot.last_message_received + 100000);
- telegram will mark all unread messages with a lower message id as read
and will remove them. Then just use the bot as normal

To save yourself a double request at the start you could just try a really
large hard coded number, or you could save the bot.last_message_recieved
value to persistent memory every time you handle a new message, and read
from there on start up

mjtrini, 2025-01-02:
Hi!

Thanks for pointing me in the right direction. The code below worked perfectly to destroy any number of messages after the Wifi connection has been established.

**//telegram shred code
secured_client.setCACert(TELEGRAM_CERTIFICATE_ROOT); // Add root certificate for api.telegram.org
int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
while (numNewMessages) {
Serial.println("shred message at boot"); 
numNewMessages = bot.getUpdates(bot.last_message_received + 1);
}**

Notice I relocated the Cert snippet. This 'shred' code in the void setup() function is perfect for eg, controlling a sliding gate motor. You don't want the gate to respond multiple times after the ESP properly boots.

Much thanks for everything!
Michael.

closed 2025-01-02 by mjtrini

witnessmenow, 2025-01-02:
Hey,

Great to hear it worked! I would increase the +1 in the while loop though
to be +100 or 1000 or something though. And probably add a small delay in
the while loop

If your bot was off for a while and say you got 50 messages, your code
would have to make 51 requests to telegram to clear out all unwanted
messages. On an Arduino each request takes roughly a second, so that might
take a minute to clear

If you increase the number you are adding on to +100, it would would only
take 2 requests to clear your queue, or 2 seconds to clear

I'm 99.99% sure that passing in that larger number will have no impact on
future messages, just ones on telegrams servers